diebietse/invertergui #16: armv7 compatibilitylibty

victor-war, 2020-09-07:
Hi, I downloaded this repository in my raspberrypi (os architecture: armv7l), built the image from the Dockerfile at https://github.com/diebietse/invertergui/ and got this error

```bash
pi@device:~ $ docker run -it --device /dev/ttyUSB2:/dev/ttyUSB0 -p 8080:8080 invertergui
INFO[0000] Starting invertergui                          ctx=inverter-gui
INFO[0000] Invertergui web server starting on: :8080     ctx=inverter-gui
2020/09/07 17:38:57 Locked
2020/09/07 17:38:59 Get scaling factors.
panic: runtime error: index out of range [2] with length 2

goroutine 23 [running]:
github.com/diebietse/invertergui/mk2driver.(*mk2Ser).scaleDecode(0x1d14080, 0x1d80002, 0x2, 0xfe)
	/build/mk2driver/mk2.go:182 +0x1a8
github.com/diebietse/invertergui/mk2driver.(*mk2Ser).handleFrame(0x1d14080, 0x1c9ed03, 0x1d80000, 0x4, 0x100)
	/build/mk2driver/mk2.go:139 +0x16c
github.com/diebietse/invertergui/mk2driver.(*mk2Ser).frameLocker(0x1d14080)
	/build/mk2driver/mk2.go:66 +0x2ec
created by github.com/diebietse/invertergui/mk2driver.NewMk2Connection
	/build/mk2driver/mk2.go:40 +0x158

```
(btw i'm using /dev/ttyUSB2:/dev/ttyUSB0 since my mk2 is connected to /dev/ttyUSB2). 

I'm asumming i'ts an architecture compatailibty problem, since this container image was meant for amd64 and i'm using armv7l, and because of that, it cant decode the values correctly.

Is there a way to make it work? maybe changing the hex values it's using on the decode part might help, but i'm clueless on how to do it.

ncthompson, 2020-09-07:
This could be a bug in the reading of the values. Maybe an assumption I made with the querying of the data. I have not tried it on a pi myself but have one lying around so will try to run it. Was this just on using rasbian?

victor-war, 2020-09-07:
Yes, it's using raspbian.
If there's anything we can do to help debug this, let us know ;).

ncthompson, 2020-09-09:
I looked at the protocol again and it seems like it is trying to read scaling factors for something that it does not have. It locks onto a valid frame and the CRC for it checks out. so there is a bug in the code as it assumes a frame length. What victron device is it trying to monitor and how new is it? It could be that the newer protocols are slightly different, that I don't take into account.

victor-war, 2020-09-09:
it's a victron multiplus II communicating via mk3 connector.
btw... we've used this documentation in the past to get victron devices data https://drive.google.com/file/d/1izXkZv-RjrY74rY1v6kb0eqikMHC6-Fm/view hope it helps.

ncthompson, 2020-09-09:
I have a branch with fixes they could work. It does not break on my side, but I would need you to test it on your system: https://github.com/diebietse/invertergui/tree/scale-factor-fixes

victor-war, 2020-09-10:
Hi,
I rebuilt the image again and got this

```bash
pi@sunai_casa_pi:~/invertergui $ docker run -it --device /dev/ttyUSB2:/dev/ttyUSB0 -p 8080:8080 invertergui
INFO[0000] Starting invertergui                          ctx=inverter-gui
INFO[0000] Invertergui web server starting on: :8080     ctx=inverter-gui
2020/09/10 21:32:30 Locked
2020/09/10 21:32:31 Get scaling factors.
2020/09/10 21:32:32 Skiping scaling factors for: 10
2020/09/10 21:32:33 Monitoring starting.
panic: runtime error: index out of range [13] with length 13

goroutine 3 [running]:
github.com/diebietse/invertergui/mk2driver.(*mk2Ser).applyScale(...)
        /build/mk2driver/mk2.go:251
github.com/diebietse/invertergui/mk2driver.(*mk2Ser).stateDecode(0x145a5c0, 0x14e8102, 0x6, 0xfe)
        /build/mk2driver/mk2.go:305 +0x84
github.com/diebietse/invertergui/mk2driver.(*mk2Ser).handleFrame(0x145a5c0, 0x1410307, 0x14e8100, 0x8, 0x100)
        /build/mk2driver/mk2.go:162 +0x198
github.com/diebietse/invertergui/mk2driver.(*mk2Ser).frameLocker(0x145a5c0)
        /build/mk2driver/mk2.go:87 +0x2ec
created by github.com/diebietse/invertergui/mk2driver.NewMk2Connection
        /build/mk2driver/mk2.go:61 +0x158
```

Are you also using a victron multiplus II? maybe our device it's not returning the same output format as yours.
How could we check this? I can make some changes locally in the go code to check.

ncthompson, 2020-09-11:
No I have one of the V1 multiplus inverters. I forgot to append a scaling factor to the list when it is not avalible, so that is why it is crashing now. I have made an update to the branch, so you can pull it again and build locally. It looks promising because from your logs it seems that the multiplus II does not have virtual switch, so it should skip that now.

victor-war, 2020-09-11:
it deployed correctly!!
there's a mismatch  in the values like 12220% battery storage and some other stuff. will upload a screenshot in another issue, (we'll try to figure it out on our own, but just so you know about the bug).

btw thanks for your help, we've made great advance in monitoring our pvsystem thanks to you, we could even get a connection from the device before you started making these fixes.

closed 2020-09-13 by ncthompson