diebietse/invertergui #21: Docker error

cairanvanrooyen, 2020-10-08:
Hi,

I have purchased a [Victron inverter/charger (MultiPlus 1600VA, 12V Inverter Charger - 1.3kW)][1] and their [MK3 USB interface (VE.Bus to USB)][2]. I am trying to get your super nice GUI to work with my Multiplus and RPi 4.

I am an experienced python (data science) user, but new to RPi and other programming languages, so apologies if this is a really basic problem.

I have cloned the repository and installed all of the dependencies (i think), then I get stuck at the following and dont know where to go from here:

```
    pi@raspberrypi:~/invertergui $ docker run --name invertergui --device /dev/ttyUSB0:/dev/ttyUSB0 -p 8080:8080 diebietse/invertergui
    docker: Error response from daemon: Conflict. The container name "/invertergui" is already in use by container "d0adb7a565d8e3cd6e80a6bac6850e2357016be4bc76b3488f3283f7ffa984e6". You have to remove (or rename) that container to be able to reuse that name.
    See 'docker run --help'.
```

  [1]: https://www.victronenergy.com/inverters-chargers/multi-500-va
  [2]: https://www.victronenergy.com/accessories/interface-mk3-usb#downloads
  [3]: https://docs.victronenergy.com/victronconnect-vebus.html#warning

ncthompson, 2020-10-08:
This looks like there is already a docker with the name `invertergui` on your system. Running `docker ps --all` should show you running and non-running containers. You should be able to delete it with  `docker rm d0adb7a565d8e3cd6e80a6bac6850e2357016be4bc76b3488f3283f7ffa984e6`. You can also run your docker command with the `--rm` flag,  that it can cleanup after itself if you are done with it (`docker run --name invertergui --rm --device /dev/ttyUSB0:/dev/ttyUSB0 -p 8080:8080 diebietse/invertergui`

cairanvanrooyen, 2020-10-08:
> This looks like there is already a docker with the name `invertergui` on your system. Running `docker ps --all` should show you running and non-running containers. You should be able to delete it [...]

Thank you so much, i will give this a go shortly. Once I have the docker running, what commands do i need to issue to get through to the GUI dashboard?

ncthompson, 2020-10-08:
Just navigate to `http://localhost:8080` on a browser if you are doing from you pi, or `http://your-pi-ip-address:8080` if you are doing it from another machine

cairanvanrooyen, 2020-10-08:
So, i have successfully removed the docker and reinstated it as you suggested, but it's showing a further error:

```
pi@raspberrypi:~ $ docker ps --all
CONTAINER ID        IMAGE                   COMMAND             CREATED             STATUS                    PORTS               NAMES
d0adb7a565d8        diebietse/invertergui   "./invertergui"     23 hours ago        Exited (1) 23 hours ago                       invertergui
79d811d16fc7        hello-world             "/hello"            23 hours ago        Exited (0) 23 hours ago                       flamboyant_poitras
pi@raspberrypi:~ $ docker rm d0adb7a565d8e3cd6e80a6bac6850e2357016be4bc76b3488f3283f7ffa984e6
d0adb7a565d8e3cd6e80a6bac6850e2357016be4bc76b3488f3283f7ffa984e6
pi@raspberrypi:~ $ docker ps --all
CONTAINER ID        IMAGE               COMMAND             CREATED             STATUS                    PORTS               NAMES
79d811d16fc7        hello-world         "/hello"            23 hours ago        Exited (0) 23 hours ago                       flamboyant_poitras
pi@raspberrypi:~ $ docker run --name invertergui --rm --device /dev/ttyUSB0:/dev/ttyUSB0 -p 8080:8080 diebietse/invertergui
standard_init_linux.go:211: exec user process caused "exec format error"
```

When i try to navigate to the webpage, i get the "This site can’t be reached"

ncthompson, 2020-10-08:
It looks like you are running into an architecture issue. We only publish x86 docker images at the moment, so if you are running this on a pi you will have to build the docker image yourself on that device and run it.

cairanvanrooyen, 2020-10-08:
Ah :(

Please can you point me in the right direction for simply grabbing data from the USB interface - ideally using Python? 

I really only need a way to get the data, and from here I can manipulate myself.

ncthompson, 2020-10-08:
You can just clone the repo on the PI and run `docker build --tag invertergui .` and then `docker run --name invertergui --device /dev/ttyUSB0:/dev/ttyUSB0 -p 8080:8080 invertergui` and it should just work.

If you want to do it yourself, you can look at the old code this is based on. https://github.com/ncthompson/inverter_monitor

cairanvanrooyen, 2020-10-08:
Thank you so much!! 

I have got the really nice Grafana graphics working!

I know there are many other Victron users who would be interested in this, so i will try and share my experiences and what I have learned, so they might also be able to use this great platform you have created.

ncthompson, 2020-10-08:
Glad it worked for you!

closed 2020-10-08 by ncthompson