diebietse/invertergui #23: Cant build docker image on latest master

GregoryOrciuch, 2021-05-06:
Hi,

just pulled latest master here. previously was using branch for 'scalefixes', now I saw it was merged so I switched back to master.
However now I cannot build:

```
root@raspberrypi:~/invertergui# git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
root@raspberrypi:~/invertergui# docker build /root/invertergui/
Sending build context to Docker daemon  16.21MB
Step 1/14 : FROM golang:alpine as builder
 ---> fb7d6f0a68ad
Step 2/14 : RUN apk add git
 ---> Using cache
 ---> 30d341c7caec
Step 3/14 : RUN mkdir /build
 ---> Using cache
 ---> 0255b830279c
Step 4/14 : COPY . /build/
 ---> 5cc8f5b0e029
Step 5/14 : WORKDIR /build
 ---> Running in 4591237d2922
Removing intermediate container 4591237d2922
 ---> c082a678ea25
Step 6/14 : RUN go build -o invertergui ./cmd/invertergui
 ---> Running in 7fee8472f07a
plugins/webui/static/binary.go:4:2: cannot find package "." in:
	/build/vendor/embed
The command '/bin/sh -c go build -o invertergui ./cmd/invertergui' returned a non-zero code: 1
```

any idea?

ncthompson, 2021-05-06:
My first guess would be that the version of `golang:alpine` that is cached on your raspberrypi is old. Try running `docker rmi golang:alpine` to remove it and build again if possible.

GregoryOrciuch, 2021-05-06:
Thanks,

now I get this one, but my understanding it's a communication issue and temporal, so will try again later

```
root@raspberrypi:~/invertergui# docker build .
Sending build context to Docker daemon  16.21MB
Step 1/14 : FROM golang:alpine as builder
 ---> de84cba6178a
Step 2/14 : RUN apk add git
 ---> Running in fad0863fbe89
fetch https://dl-cdn.alpinelinux.org/alpine/v3.13/main/armv7/APKINDEX.tar.gz
ERROR: https://dl-cdn.alpinelinux.org/alpine/v3.13/main: temporary error (try again later)
WARNING: Ignoring https://dl-cdn.alpinelinux.org/alpine/v3.13/main: No such file or directory
fetch https://dl-cdn.alpinelinux.org/alpine/v3.13/community/armv7/APKINDEX.tar.gz
ERROR: https://dl-cdn.alpinelinux.org/alpine/v3.13/community: temporary error (try again later)
WARNING: Ignoring https://dl-cdn.alpinelinux.org/alpine/v3.13/community: No such file or directory
ERROR: unable to select packages:
  git (no such package):
    required by: world[git]
The command '/bin/sh -c apk add git' returned a non-zero code: 1
```

GregoryOrciuch, 2021-05-06:
this is weird, host can download that files, but docker during build does not want to do so, any idea ?

GregoryOrciuch, 2021-05-06:
I've found that latest alpine 3.13 has some problems with dns  stuff. Changing builder image to
FROM golang:alpine3.12

solves the problem.
(otherwise it picks up the latest 3.13)

closed 2021-05-06 by GregoryOrciuch

ncthompson, 2021-05-06:
Good to know, thanks for the feedback.

ncthompson, 2021-05-06:
We now build images for arm on github, for our latest release you can pull the images instead of building it. 
Try the docker image; `ghcr.io/diebietse/invertergui:2.1.0`