diebietse/invertergui #27: Incorrect output current and voltage

Banshee1221, 2021-12-08:
Firstly, let me state that this project is awesome. I am excited about collecting the metrics with Prometheus.

I have an issue with my Victron MultiPlus 24V/3000 using this software. It seems out output the incorrect current and voltage values for outputs when the system is not under load. I am using a MK2 USB adapter.

Example is how it normally looks when not under load:
[image]

But every few refreshes it will give the correct value:
[image]

When placed under load, it seems to work just fine. Is this something you've seen before?

Thanks

ncthompson, 2021-12-08:
Hey, glad you are enjoying this project. This looks like it could be a scaling issue. Could you maybe provide us with the version you are using? It would also be very helpful if you can run the service in debug mode and provide us with the log output for when you see these values. --loglevel=debug should do it.

Banshee1221, 2021-12-08:
Hi @ncthompson thanks for responding.

Currently, I've actually forked the project and build a docker image from master for arm32v6 (Pi Zero W). However, the same occurred when I was using ghcr.io/diebietse/invertergui:latest on my laptop for testing, so it's definitely not to do with the architecture change.

Here are logs from the master build (they look the same on x86 and arm32v6). You will see here that one response has the correct values and the second has the whacky values.

```
time="2021-12-08T14:49:47Z" level=debug msg="frame []byte{0xff, 0x56, 0x98, 0x3e, 0x11, 0x0, 0x0, 0xbd}"
time="2021-12-08T14:49:47Z" level=debug msg="versiondecode [152 62 17 0 0 189]"
time="2021-12-08T14:49:47Z" level=debug msg="sendCommand []byte{0x3, 0xff, 0x46, 0x0, 0xb8}"
time="2021-12-08T14:49:47Z" level=debug msg="frame []byte{0x20, 0xb6, 0x89, 0x6d, 0xb7, 0xc, 0x51, 0xa, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x87, 0x80}"
time="2021-12-08T14:49:47Z" level=debug msg="dcDecode &mk2driver.Mk2Info{Valid:true, Version:0xac0, BatVoltage:26.41, BatCurrent:0, InVoltage:0, InCurrent:0, InFrequency:0, OutVoltage:0, OutCurrent:0, OutFrequency:50.153452685421996, ChargeState:0, LEDs:map[mk2driver.Led]mk2driver.LEDstate(nil), Errors:[]error(nil), Timestamp:time.Time{wall:0x0, ext:0, loc:(*time.Location)(nil)}}"
time="2021-12-08T14:49:47Z" level=debug msg="sendCommand []byte{0x3, 0xff, 0x46, 0x1, 0xb7}"
time="2021-12-08T14:49:47Z" level=debug msg="frame []byte{0x20, 0x1, 0x1, 0x6d, 0xb7, 0x8, 0x70, 0x5c, 0x21, 0x0, 0x70, 0x5c, 0x0, 0x0, 0xc3, 0x27}"
time="2021-12-08T14:49:47Z" level=debug msg="acDecode &mk2driver.Mk2Info{Valid:true, Version:0xac0, BatVoltage:26.41, BatCurrent:0, InVoltage:236.64000000000001, InCurrent:0.33, InFrequency:50.1025641025641, OutVoltage:236.64000000000001, OutCurrent:0, OutFrequency:50.153452685421996, ChargeState:0, LEDs:map[mk2driver.Led]mk2driver.LEDstate(nil), Errors:[]error(nil), Timestamp:time.Time{wall:0x0, ext:0, loc:(*time.Location)(nil)}}"
---
time="2021-12-08T14:49:49Z" level=debug msg="frame []byte{0xff, 0x56, 0x98, 0x3e, 0x11, 0x0, 0x0, 0xbd}"
time="2021-12-08T14:49:49Z" level=debug msg="versiondecode [152 62 17 0 0 189]"
time="2021-12-08T14:49:49Z" level=debug msg="sendCommand []byte{0x3, 0xff, 0x46, 0x0, 0xb8}"
time="2021-12-08T14:49:49Z" level=debug msg="frame []byte{0x20, 0xb6, 0x89, 0x6d, 0xb7, 0xc, 0x51, 0xa, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x88, 0x7f}"
time="2021-12-08T14:49:49Z" level=debug msg="dcDecode &mk2driver.Mk2Info{Valid:true, Version:0xac0, BatVoltage:26.41, BatCurrent:0, InVoltage:0, InCurrent:0, InFrequency:0, OutVoltage:0, OutCurrent:0, OutFrequency:50.025510204081634, ChargeState:0, LEDs:map[mk2driver.Led]mk2driver.LEDstate(nil), Errors:[]error(nil), Timestamp:time.Time{wall:0x0, ext:0, loc:(*time.Location)(nil)}}"
time="2021-12-08T14:49:49Z" level=debug msg="sendCommand []byte{0x3, 0xff, 0x46, 0x1, 0xb7}"
time="2021-12-08T14:49:49Z" level=debug msg="frame []byte{0x20, 0x1, 0x1, 0x6d, 0xb7, 0x8, 0x70, 0x5c, 0x21, 0x0, 0x70, 0x5c, 0xfe, 0xff, 0xc3, 0x2a}"
time="2021-12-08T14:49:49Z" level=debug msg="acDecode &mk2driver.Mk2Info{Valid:true, Version:0xac0, BatVoltage:26.41, BatCurrent:0, InVoltage:236.64000000000001, InCurrent:0.33, InFrequency:50.1025641025641, OutVoltage:236.64000000000001, OutCurrent:655.34, OutFrequency:50.025510204081634, ChargeState:0, LEDs:map[mk2driver.Led]mk2driver.LEDstate(nil), Errors:[]error(nil), Timestamp:time.Time{wall:0x0, ext:0, loc:(*time.Location)(nil)}}"
```

I've compiled 2.1.1 now for the Pi Zero and I see similar behaviour, but it seems to have the right values more often than master:

```
time="2021-12-08T15:04:45Z" level=info msg="Starting invertergui" ctx=inverter-gui
time="2021-12-08T15:04:45Z" level=debug msg="sendCommand []byte{0x4, 0xff, 0x41, 0x1, 0x0, 0xbb}"
time="2021-12-08T15:04:45Z" level=info msg="Invertergui web server starting on: :8080" ctx=inverter-gui
time="2021-12-08T15:04:45Z" level=info msg=Locked
time="2021-12-08T15:04:46Z" level=debug msg="frame []byte{0xff, 0x56, 0x98, 0x3e, 0x11, 0x0, 0x0, 0xbd}"
time="2021-12-08T15:04:46Z" level=debug msg="versiondecode [152 62 17 0 0 189]"
time="2021-12-08T15:04:46Z" level=info msg="Get scaling factors."
time="2021-12-08T15:04:46Z" level=debug msg="sendCommand []byte{0x5, 0xff, 0x57, 0x36, 0x0, 0x0, 0x6f}"
time="2021-12-08T15:04:46Z" level=debug msg="frame []byte{0xff, 0x57, 0x8e, 0x9c, 0x7f, 0x8f, 0x0, 0x0, 0x6a}"
time="2021-12-08T15:04:46Z" level=debug msg="Scale frame(7): 0x8e9c7f8f00006a"
time="2021-12-08T15:04:46Z" level=debug msg="scalecount 0: mk2driver.scaling{scale:0.01, offset:0, signed:false, supported:true} \n"
time="2021-12-08T15:04:46Z" level=debug msg="sendCommand []byte{0x5, 0xff, 0x57, 0x36, 0x1, 0x0, 0x6e}"
time="2021-12-08T15:04:46Z" level=debug msg="frame []byte{0xff, 0x57, 0x8e, 0x9c, 0x7f, 0x8f, 0x0, 0x0, 0x6a}"
time="2021-12-08T15:04:46Z" level=debug msg="Scale frame(7): 0x8e9c7f8f00006a"
time="2021-12-08T15:04:46Z" level=debug msg="scalecount 1: mk2driver.scaling{scale:0.01, offset:0, signed:false, supported:true} \n"
time="2021-12-08T15:04:46Z" level=debug msg="sendCommand []byte{0x5, 0xff, 0x57, 0x36, 0x2, 0x0, 0x6d}"
time="2021-12-08T15:04:46Z" level=debug msg="frame []byte{0xff, 0x57, 0x8e, 0x9c, 0x7f, 0x8f, 0x0, 0x0, 0x6a}"
time="2021-12-08T15:04:46Z" level=debug msg="Scale frame(7): 0x8e9c7f8f00006a"
time="2021-12-08T15:04:46Z" level=debug msg="scalecount 2: mk2driver.scaling{scale:0.01, offset:0, signed:false, supported:true} \n"
[... cut ...]
time="2021-12-08T15:04:59Z" level=debug msg="frame []byte{0xff, 0x57, 0x85, 0xc8, 0x0, 0x58}"
time="2021-12-08T15:04:59Z" level=debug msg="battery state decode &mk2driver.Mk2Info{Valid:true, Version:0xac0, BatVoltage:26.38, BatCurrent:0, InVoltage:234.98000000000002, InCurrent:0.33, InFrequency:50.1025641025641, OutVoltage:234.98000000000002, OutCurrent:655.32, OutFrequency:50.025510204081634, ChargeState:1, LEDs:map[mk2driver.Led]mk2driver.LEDstate{0:1, 1:0, 2:0, 3:1, 4:0, 5:0, 6:0, 7:0}, Errors:[]error(nil), Timestamp:time.Time{wall:0x0, ext:0, loc:(*time.Location)(nil)}}"
```

hpdvanwyk, 2021-12-08:
Hi @Banshee1221.  

It looks like I incorrectly interpreted some signedness values to also apply to AC current. This lead to signed integers being interpreted as unsigned so a very small negative current ready read as a very large positive one. I have attempted a fix on the https://github.com/diebietse/invertergui/tree/fix-signedness branch. I don't have access to my Victron at the moment so I could only test against your debug values and our existing unit tests. 

Could you give the branch a try and see if it works for you?

Banshee1221, 2021-12-09:
Hi @hpdvanwyk 

I appears that may have fixed it. I haven't run it for too long, but it seems like it's working as expected. Here are the screenshots of what I see now...

Negative values:
[image]

Positive values:
[image]

And logs:

```
energypi:~# podman run --rm --name invertergui --device /dev/ttyUSB0:/dev/ttyUSB0 -p 8080:8080 localhost/invertergui:fixed-signedness --loglevel=debug
time="2021-12-09T06:18:10Z" level=info msg="Starting invertergui" ctx=inverter-gui
time="2021-12-09T06:18:10Z" level=debug msg="sendCommand []byte{0x4, 0xff, 0x41, 0x1, 0x0, 0xbb}"
time="2021-12-09T06:18:10Z" level=info msg="Invertergui web server starting on: :8080" ctx=inverter-gui
time="2021-12-09T06:18:10Z" level=info msg=Locked
time="2021-12-09T06:18:12Z" level=debug msg="frame []byte{0xff, 0x56, 0x98, 0x3e, 0x11, 0x0, 0x0, 0xbd}"
time="2021-12-09T06:18:12Z" level=debug msg="versiondecode [152 62 17 0 0 189]"
time="2021-12-09T06:18:12Z" level=info msg="Get scaling factors."
time="2021-12-09T06:18:12Z" level=debug msg="sendCommand []byte{0x5, 0xff, 0x57, 0x36, 0x0, 0x0, 0x6f}"
time="2021-12-09T06:18:12Z" level=debug msg="frame []byte{0xff, 0x57, 0x8e, 0x9c, 0x7f, 0x8f, 0x0, 0x0, 0x6a}"
time="2021-12-09T06:18:12Z" level=debug msg="Scale frame(7): 0x8e9c7f8f00006a"
time="2021-12-09T06:18:12Z" level=debug msg="scalecount 0: mk2driver.scaling{scale:0.01, offset:0, signed:false, supported:true} \n"
time="2021-12-09T06:18:12Z" level=debug msg="sendCommand []byte{0x5, 0xff, 0x57, 0x36, 0x1, 0x0, 0x6e}"
time="2021-12-09T06:18:12Z" level=debug msg="frame []byte{0xff, 0x57, 0x8e, 0x9c, 0x7f, 0x8f, 0x0, 0x0, 0x6a}"
time="2021-12-09T06:18:12Z" level=debug msg="Scale frame(7): 0x8e9c7f8f00006a"
time="2021-12-09T06:18:12Z" level=debug msg="scalecount 1: mk2driver.scaling{scale:0.01, offset:0, signed:false, supported:true} \n"
time="2021-12-09T06:18:12Z" level=debug msg="sendCommand []byte{0x5, 0xff, 0x57, 0x36, 0x2, 0x0, 0x6d}"
time="2021-12-09T06:18:12Z" level=debug msg="frame []byte{0xff, 0x57, 0x8e, 0x9c, 0x7f, 0x8f, 0x0, 0x0, 0x6a}"
time="2021-12-09T06:18:12Z" level=debug msg="Scale frame(7): 0x8e9c7f8f00006a"
time="2021-12-09T06:18:12Z" level=debug msg="scalecount 2: mk2driver.scaling{scale:0.01, offset:0, signed:false, supported:true} \n"
time="2021-12-09T06:18:12Z" level=debug msg="sendCommand []byte{0x5, 0xff, 0x57, 0x36, 0x3, 0x0, 0x6c}"
time="2021-12-09T06:18:12Z" level=debug msg="frame []byte{0xff, 0x57, 0x8e, 0x9c, 0x7f, 0x8f, 0x0, 0x0, 0x6a}"
time="2021-12-09T06:18:12Z" level=debug msg="Scale frame(7): 0x8e9c7f8f00006a"
time="2021-12-09T06:18:12Z" level=debug msg="scalecount 3: mk2driver.scaling{scale:0.01, offset:0, signed:false, supported:true} \n"
time="2021-12-09T06:18:12Z" level=debug msg="sendCommand []byte{0x5, 0xff, 0x57, 0x36, 0x4, 0x0, 0x6b}"
time="2021-12-09T06:18:12Z" level=debug msg="frame []byte{0xff, 0x57, 0x8e, 0x9c, 0x7f, 0x8f, 0x0, 0x0, 0x6a}"
time="2021-12-09T06:18:12Z" level=debug msg="Scale frame(7): 0x8e9c7f8f00006a"
time="2021-12-09T06:18:12Z" level=debug msg="scalecount 4: mk2driver.scaling{scale:0.01, offset:0, signed:false, supported:true} \n"
time="2021-12-09T06:18:12Z" level=debug msg="sendCommand []byte{0x5, 0xff, 0x57, 0x36, 0x5, 0x0, 0x6a}"
time="2021-12-09T06:18:12Z" level=debug msg="frame []byte{0xff, 0x57, 0x8e, 0x64, 0x80, 0x8f, 0x0, 0x0, 0xa1}"
time="2021-12-09T06:18:12Z" level=debug msg="Scale frame(7): 0x8e64808f0000a1"
time="2021-12-09T06:18:12Z" level=debug msg="scalecount 5: mk2driver.scaling{scale:0.01, offset:0, signed:true, supported:true} \n"
time="2021-12-09T06:18:12Z" level=debug msg="sendCommand []byte{0x5, 0xff, 0x57, 0x36, 0x6, 0x0, 0x69}"
time="2021-12-09T06:18:12Z" level=debug msg="frame []byte{0xff, 0x57, 0x8e, 0x9c, 0x7f, 0x8f, 0x0, 0x0, 0x6a}"
time="2021-12-09T06:18:12Z" level=debug msg="Scale frame(7): 0x8e9c7f8f00006a"
time="2021-12-09T06:18:12Z" level=debug msg="scalecount 6: mk2driver.scaling{scale:0.01, offset:0, signed:false, supported:true} \n"
time="2021-12-09T06:18:12Z" level=debug msg="sendCommand []byte{0x5, 0xff, 0x57, 0x36, 0x7, 0x0, 0x68}"
time="2021-12-09T06:18:13Z" level=debug msg="frame []byte{0xff, 0x57, 0x8e, 0x57, 0x78, 0x8f, 0x0, 0x1, 0xb5}"
time="2021-12-09T06:18:13Z" level=debug msg="Scale frame(7): 0x8e57788f0001b5"
time="2021-12-09T06:18:13Z" level=debug msg="scalecount 7: mk2driver.scaling{scale:0.0005099439061703213, offset:256, signed:false, supported:true} \n"
time="2021-12-09T06:18:13Z" level=debug msg="sendCommand []byte{0x5, 0xff, 0x57, 0x36, 0x8, 0x0, 0x67}"
time="2021-12-09T06:18:13Z" level=debug msg="frame []byte{0xff, 0x57, 0x8e, 0x2f, 0x7c, 0x8f, 0x0, 0x0, 0xda}"
time="2021-12-09T06:18:13Z" level=debug msg="Scale frame(7): 0x8e2f7c8f0000da"
time="2021-12-09T06:18:13Z" level=debug msg="scalecount 8: mk2driver.scaling{scale:0.0010235414534288639, offset:0, signed:false, supported:true} \n"
time="2021-12-09T06:18:13Z" level=debug msg="sendCommand []byte{0x5, 0xff, 0x57, 0x36, 0x9, 0x0, 0x66}"
time="2021-12-09T06:18:13Z" level=debug msg="frame []byte{0xff, 0x57, 0x8e, 0x64, 0x80, 0x8f, 0x0, 0x0, 0xa1}"
time="2021-12-09T06:18:13Z" level=debug msg="Scale frame(7): 0x8e64808f0000a1"
time="2021-12-09T06:18:13Z" level=debug msg="scalecount 9: mk2driver.scaling{scale:0.01, offset:0, signed:true, supported:true} \n"
time="2021-12-09T06:18:13Z" level=debug msg="sendCommand []byte{0x5, 0xff, 0x57, 0x36, 0xa, 0x0, 0x65}"
time="2021-12-09T06:18:13Z" level=debug msg="frame []byte{0xff, 0x57, 0x8e, 0x4, 0x0, 0x8f, 0x0, 0x80, 0x1}"
[... cut ...]
time="2021-12-09T06:18:20Z" level=debug msg="battery state decode &mk2driver.Mk2Info{Valid:true, Version:0xac0, BatVoltage:26.41, BatCurrent:0, InVoltage:236.64000000000001, InCurrent:0.33, InFrequency:50.1025641025641, OutVoltage:236.64000000000001, OutCurrent:0, OutFrequency:49.77157360406091, ChargeState:1, LEDs:map[mk2driver.Led]mk2driver.LEDstate{0:1, 1:0, 2:0, 3:1, 4:0, 5:0, 6:0, 7:0}, Errors:[]error(nil), Timestamp:time.Time{wall:0x0, ext:0, loc:(*time.Location)(nil)}}"
```

I guess we can probably close this issue after merging to master and I'll re-open it if I notice any oddity?

Thanks so much for getting to this so quickly :)

ncthompson, 2021-12-09:
Good to see it working. Just a question on using it on your Pi. We build the arm v7 Docker image so you should be able to pull the image directly, or do you use the binary?

Banshee1221, 2021-12-09:
Hello @ncthompson,

I use the Pi Zero W (first gen) for monitoring the inveter, it uses ARM v6, not v7 so the Docker image is not compatible.

Would you mind me making a pull request to add ARMv6 support? (I'm sure I'm one of the very few people that will ever use the Pi Zero for this, but maybe it helps some others?)

ncthompson, 2021-12-09:
Please do!

closed 2022-01-25 by ncthompson