diebietse/invertergui #33: Allow remote Victron ON/OFF

sanitariu, 2022-05-04:
Hello,
is it possible to pass some flag 1/0 to force victron multiplus stop using batteries and later start using batteries ? The idea 
is to make it remove switch to AC power and return back to batteries.

ncthompson, 2022-06-23:
The idea with this project is to serve as only a monitor into your Victron system. We do not want to add functionality that can change the state of a system as this can cause harm to you or your system if we get this wrong. I would suggest using Victron's provides Venus OS (https://github.com/victronenergy/venushttps://github.com/victronenergy/venus) if you want to control your inverter.

closed 2022-06-23 by ncthompson

WanderingBison, 2022-06-23:
While I fully agree with your rationale, one issue that Victron is aware of is the inability to control the state of the inverter through Venus OS when also using their BMS and batteries. It appears to be a hardware issue. 

That would be one reason to add state control to invertergui. 

Thanks.