diebietse/invertergui #38: IP (NS) protection log access

DurvalMenezes, 2023-03-29:
Is there a way (or is it possible to add one) for invertergui to read events from a Victron inverter "NS protection log" (please see attached screen). 
 
[image]

This is available on VEConfig3 under the "Grid" tab by pressing the "Show IP (NS) protection log", and is very very useful to diagnose why the heck the inverter turned its mains input off (usually because it went out of acceptable voltage and/or frequency range).  I use it all the time while I'm tuning my new generator and the inverter config to work smoothly with each other, and it's a PITA to take the cable out of my RPi4 (where I run invertergui) and plug it into my PC where I have VEConfig3 installed. 

It would be great if invertergui could monitor this log either periodically (like once a minute) or on demand, and dump any new events either to stdout or (even better) log it to syslog and/or MQTT.

Please let me know if I can be of any help, for example capturing traffic on the MK3 USB port when the button is pressed in VEConfig3.

ncthompson, 2023-03-30:
I had a quick look at the docs. It looks like there is a section on alarms raised, but I am not sure if this is generated the same way. It could help if you can capture the traffic. I am mostly interested in the serial comms over USB.

DurvalMenezes, 2023-03-30:
Thanks for the prompt response. I run VEConfig3 inside a VirtualBox VM, which can be set to capture all USB traffic to a pcap-formatted file which can then be decoded with Wireshark. I'm pretty confident that Wireshark will allow me to extract the embedded serial comms data.

I will give it a try the next time I have the MK3 interface plugged to my PC, and then post the results here.

DurvalMenezes, 2023-04-01:
> I will give it a try the next time I have the MK3 interface plugged to my PC, and then post the results here.

Just did it, and I have good news and bad news: 

- The good news is that I was able to capture the traffic on a wireshark PCAPNG file, please download it here: https://durval.com/xfer-only/Victron_MK3_USB_Traffic_When_Show_IP_NS_Protection_Log_Is_Pressed.pcapng; using the Virtualbox USB traffic capture did not work well due to it being hard to start/stop the capture on the moments right before pressing the  "Show IP (NS) protection log" button and the resulting window with the log being shown on screen, so I ended up capturing it directly via Wireshark using the "usbmon" method described here: https://wiki.wireshark.org/CaptureSetup/USB

- The bad news is that I was not able to extract the embedded serial comms data; Wireshark (v.2.6.10) simply did not have a way to do that (or I could not find it). To make things worse, there was a *ton* of traffic (over 100 USB packets per second) between my PC and the MK3  device; I'm sure all traffic was to/from MK3 and not any other device because the destination bus and device (1 and 25) were exactly the one assigned to it:

```
lsusb | grep 0403:6015
    Bus 001 Device 025: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
```

- I was able to capture the packets for the time interval approximately (~1s precision) right before clicking on the "Show IP (NS) protection log" button and right after the window with the log showing up, but even so the capture file is over 500 packets long :-/

- In case it helps to make sense of the packet capture, here's the window showing the log, that came up with the above capture: 
[image]

Please let me know if I can be of any more help, or if there's a better way to capture and/or decode the traffic.

xQwexx, 2023-04-11:
I am not sure it is helpful but [I found this project ](https://github.com/martiby/multiplus2/blob/master/multiplus2.py)which has some low voltage status output I think based on the led status

DurvalMenezes, 2023-05-01:
> I am not sure it is helpful but [I found this project ](https://github.com/martiby/multiplus2/blob/master/multiplus2.py)which has some low voltage status output I think based on the led status
IP
Just had a look and unfortunately this code doesn't have any functionality for accessing the IP (NS) protection log -- and the only voltage condition it mentions is related to the battery, not to AC mains :-/ But anyway, thanks for trying to help.