diebietse/invertergui #44: Turn off/on victron stop metrics changes

jeanlucmongrain, 2024-07-22:
Victron is connected to a single-board computer through /dev/ttyUSB0.
Invetergui is running from ARM64 Docker image.

But everytime I turn off Victron for a while and turn it on. There is nore new metrics values published. But the metrics as still being ingested from the serial interface, as the debug logs show changes in values pulled from the Victron, example:

```
time="2024-07-22T16:35:29Z" level=debug msg="acDecode &mk2driver.Mk2Info{Valid:true, Version:0x1110, BatVoltage:26.84, BatCurrent:2.3000000000000003, InVoltage:0, InCurrent:0, InFrequency:0, OutVoltage:120.31, OutCurrent:0.42, OutFrequency:59.969418960244646, ChargeState:0, LEDs:map[mk2driver.Led]mk2driver.LEDstate{0:0, 1:0, 2:0, 3:0, 4:1, 5:0, 6:0, 7:0}, Errors:[]error(nil), Timestamp:time.Time{wall:0x0, ext:0, loc:(*time.Location)(nil)}}"
```

and later:

```
time="2024-07-22T16:35:31Z" level=debug msg="acDecode &mk2driver.Mk2Info{Valid:true, Version:0x1110, BatVoltage:26.85, BatCurrent:2.2, InVoltage:0, InCurrent:0, InFrequency:0, OutVoltage:120.31, OutCurrent:0.42, OutFrequency:59.969418960244646, ChargeState:0, LEDs:map[mk2driver.Led]mk2driver.LEDstate{0:0, 1:0, 2:0, 3:0, 4:1, 5:0, 6:0, 7:0}, Errors:[]error(nil), Timestamp:time.Time{wall:0x0, ext:0, loc:(*time.Location)(nil)}}"
```

Among others, the value of `BatVoltage` changed from `26.84` to `26.85`.

but if I look at prometheus metrics the value is stuck at when I restarted Victron:

```
# HELP battery_voltage_v Voltage of the battery.
# TYPE battery_voltage_v gauge
battery_voltage_v 26.79
```

jeanlucmongrain, 2024-07-22:
https://github.com/diebietse/invertergui/blob/47e73a4eff22ade6dad03d6e3926fba1f2fc8d88/plugins/prometheus/prometheus.go#L126

There is no mechanism that reset `Valid` to `True` whenever the Victron is back online, this the metrics are never updated

ncthompson, 2024-07-22:
The valid flag is set on the `versionDecode` function call https://github.com/diebietse/invertergui/blob/master/mk2driver/mk2.go#L284. Then the latest info is sent in after we receive charge state here https://github.com/diebietse/invertergui/blob/master/mk2driver/mk2.go#L383

Do you see any debug logs that contain `battery state decode` or `versiondecode`?

jeanlucmongrain, 2024-07-22:
I just rebuild from `master`, added some extra debug logging and error logging.

And the value of `.Valid` is always true going through a powercycle of the Victron. So it's not the reason.

I'm adding extra logging statements to figure

> Do you see any debug logs that contain battery state decode or versiondecode?

yes actually

jeanlucmongrain, 2024-07-22:
> Do you see any debug logs that contain battery state decode or versiondecode?

just `versiondecode`

ncthompson, 2024-07-22:
Without `stateDecode` there will be no new info update sent to prometheus. I am unsure as to why you are not receiving charge state updates.

jeanlucmongrain, 2024-07-22:
PR #45 improve things a bit, but I still don't get anymore `commandReadRAMResponse` wimon frames once it's restarted

jeanlucmongrain, 2024-07-22:
beside `setTarget()` is there anything that need to be send to Victron to make it return winmon frames?

jeanlucmongrain, 2024-07-31:
If invertergui starts while the victron is off and later turned on. the metrics are never digested.
and it will never get the winmon frame of startup if `setTarget()` is never executed.

the only way to really handle that other case as well is looping and run `setTarget()` until it works...